surfrock66/torque #1: No Sessions Available

pbuckley4192, 2016-04-21:
In the demo, 
I noticed that you can filter to sessions in  2014 however no sessions exist in 2014.

It would make more sense to only filter by the range of dates that you do have.

surfrock66, 2016-04-21:
Hmm, that's a good issue.  Let me explain.

There's a "year" filter, and a "month" filter.  If no month or year is selected, it chooses the CURRENT month/year.  The options in the drop down are populated from the possible values stored in the database.

I started recording tracks in October 2014.  So, you can choose 2014...but when you do, if you filter just on that, it's currently choosing April 2014 (since it's April 2016), where there's no tracks.

The way I have the filtering done, I'm not sure how to fix that...I guess upon choosing a year, it could re-query the months (and vice versa) and try to dynamically update the month list...I'll have to look into that.

surfrock66, 2016-04-22:
Ok, I pushed out a POTENTIAL fix.  Not sure if it's better or not.  Commit is here:

https://github.com/surfrock66/torque/commit/66bcd4dcf784e80f88f76ae7c356d6ae8aeac836

Cases:
If you choose NO filter, it shows you ANY profile, Current year, Current month.
If you filter Profile only, it shows you that profile, current year, current month.
If you filter Year only, it shows you ANY profile, ALL months for that year.
If you filter month only, it show you ANY profile, Current year, specified month.
If you filter year AND month, it filters any profile from that year and month.
If you filter profile AND year, it shows you all sessions from that profile in that year from any month.
If you filter profile AND month, it shows you sessions from that profile and that month in the current year.
If you filter All 3, it'll give you only what you want.

I don't know if that's better?  Should reduce the "no sessions" instances, unless you choose a month in the future.

pbuckley4192, 2016-04-25:
Perfect, 
when I get home later this evening,
I'll apply the patch but judging from the comment it appears to fix the issue!

closed 2016-04-25 by pbuckley4192

surfrock66, 2017-01-08:
OK, I think this is fixed in these 2 commits:

05f76ed
44bc460

The ending behavior is, if there's an empty session result set, pull the last 20 sessions with no filter.  If that doesn't fix it, I don't know what would.  It's blunt, but it should be effective.