surfrock66/torque #5: GRANT USAGE, FILE TO 'steve'@'localhost'; Error

xXAzazelXx, 2016-06-26:
Hey guys,

Im very new to this and dont know SQL :(

When i follow your guide and at very start when i try to create database I get the same error.

GRANT USAGE, FILE TO 'azazel'@'localhost';

mysql> GRANT USAGE, FILE 'azazel'@'localhost';

ERROR 1064 (42000): You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near ''azazel'@'localhost'' at line 1

Everything above worked fine.

mysql> CREATE DATABASE torque;
Query OK, 1 row affected (0.24 sec)

mysql> CREATE USER 'azazel'@'localhost' IDENTIFIED BY 'blablabla';
Query OK, 0 rows affected (0.00 sec)

Any ideas? :)

aldodemi, 2016-06-28:
for those like you, and me at my first try, that don't have any other database, try instead to use:
GRANT ALL ON *.* TO ‘azazel’@’localhost’;

xXAzazelXx, 2016-06-29:
mysql> GRANT ALL ON . TO ‘azazel’@’localhost’;
ERROR 1064 (42000): You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'TO ‘azazel’@’localhost’' at line 1

No Sorry, this is what I'm getting

aldodemi, 2016-06-29:
Sorry lost something.....
the correct command is with stars/asterisks:
GRANT ALL ON *.* TO ‘azazel’@’localhost’;

aldodemi, 2016-06-29:
don't understand why stars/asterisks are removed, try again:
"GRANT ALL ON *.* TO ‘azazel’@’localhost’;"

aldodemi, 2016-06-29:
no way to show the stars asterisks......
by the way: GRANT ALL ON "star"."star" TO ‘azazel’@’localhost’;
also look at the torque forum https://torque-bhp.com/forums/?wpforumaction=viewtopic&t=7617.0

xXAzazelXx, 2016-06-29:
mysql> GRANT ALL ON *.* TO ‘azazel’@’localhost’;
Query OK, 0 rows affected (0.00 sec)

Thank you :)

closed 2017-01-15 by surfrock66