surfrock66/torque #6: Litte help for old data recovery

aldodemi, 2016-07-06:
I've used the "econpy" version till last week, now that I've found this new version I started use it with the previous database adding the two new tables (create_torque_keys_table.sql and create_torque_log_table.sql). New trips are recorded and displayed correctly the old ones obviously no. There is a way to recover also old trips? Thank in advance for your help.

surfrock66, 2016-07-06:
Lemme take a look a little later....I can't think of a way to recover now, but I just had a baby and am in the hospital and am a little scatterbrained! 😁

aldodemi, 2016-07-06:
Just think to your baby, database and a lot of other things can wait!!!!!!!!!!!!!!!!! :-)

surfrock66, 2016-07-09:
Ok, so, I don't think there's a way to recover the data :(

This is a great point though; I just copied the table create script from econpy's repo, and it doesn't accommodate if there's a table there already.  I never really considered a migration path from his install to mine, and unfortunately you paid the price for that, and I'm sorry.

I'm gonna update the table create scripts and maybe do a rename of existing tables, instead of a drop.

surfrock66, 2016-07-09:
This commit should fix it, I tested it on a temp table:

https://github.com/surfrock66/torque/commit/f63ee7f5c0b0bd2e760e83828bb8d8fdce5b40d8

aldodemi, 2016-07-09:
No, I’m sorry, my request is different.

I’ve not used your tablescript to recreate the main database, so my previous data are still in the database.

My problem is, simply, that all “old” record doesnt show, I think because they dont have a link in the key table, so my question was if there is a way to create missing information for these old record.

Thank

Aldo

surfrock66, 2016-07-09:
Oh!  My mistake, I fully admit to being a bit distracted.  That's good that your data is still there.

Rather than having a separate keys table; econpy stored the field info as a comment in the raw_logs table.  Run this in the mysql console to see the full list of info:

`use torque; show full columns from raw_logs;`

From there, you need to compare that to the output of this:

`use torque; show full columns from raw_logs;`

raw_logs.Field => torque_keys.id
raw_logs.Comment => torque_keys.description
raw_logs.Type => torque_keys.type

The "torque_keys.units", "torque_keys.min", and "torque_keys.max" variables are not super important; I'll dump mine below since they should be pretty universal.

You'll need to do an insert query into the "torque_keys" table for any variables that have a column in raw_logs, but aren't in torque_keys.

The big one here though, is "torque_keys.populated".  If that's a 0, it'll not show up in the variable select list...this is my way of logging everything but only graphing relevant data.  You can edit this in the mysql console with an update query, or from the /torque/pid_edit.php page.

Does that help?

```
mysql> use torque; select * from torque_keys;
Database changed
+-----------------+--------------------------------------------------+--------------+---------+-----------+------+-------+
| id              | description                                      | type         | units   | populated | min  | max   |
+-----------------+--------------------------------------------------+--------------+---------+-----------+------+-------+
| kff122e         | 0-100kph Time                                    | float        | s       |         0 |    0 |    16 |
| kff124f         | 0-200kph Time                                    | float        | s       |         0 |    0 |    16 |
| kff1277         | 0-30mph Time                                     | float        | s       |         0 |    0 |    16 |
| kff122d         | 0-60mph Time                                     | float        | s       |         0 |    0 |    16 |
| kff122f         | 1/4 mile Time                                    | float        | s       |         0 |    0 |    30 |
| kff1230         | 1/8 mile Time                                    | float        | s       |         0 |    0 |    16 |
| kff1264         | 100-0kph Time                                    | float        | s       |         0 |    0 |    16 |
| kff1260         | 40-60mph Time                                    | float        | s       |         0 |    0 |    16 |
| kff1265         | 60-0mph Time                                     | float        | s       |         0 |    0 |    16 |
| kff125e         | 60-120mph Time                                   | float        | s       |         0 |    0 |    16 |
| kff1276         | 60-130mph Time                                   | float        | s       |         0 |    0 |    16 |
| kff125f         | 60-80mph Time                                    | float        | s       |         0 |    0 |    16 |
| kff1261         | 80-100mph Time                                   | float        | s       |         0 |    0 |    16 |
| kff1275         | 80-120kph Time                                   | float        | s       |         0 |    0 |    16 |
| k47             | Absolute Throttle Position B                     | float        | %       |         1 |    0 |   100 |
| kff1223         | Acceleration Sensor (Total)                      | float        | g       |         0 |   -1 |     1 |
| kff1220         | Acceleration Sensor (X Axis)                     | float        | g       |         1 |   -1 |     1 |
| kff1221         | Acceleration Sensor (Y Axis)                     | float        | g       |         1 |   -1 |     1 |
| kff1222         | Acceleration Sensor (Z Axis)                     | float        | g       |         1 |   -1 |     1 |
| k49             | Accelerator Pedal Position D                     | float        | %       |         1 |    0 |   100 |
| k4a             | Accelerator Pedal Position E                     | float        | %       |         1 |    0 |   100 |
| k4b             | Accelerator Pedal Position F                     | float        | %       |         0 |    0 |   100 |
| kff124d         | Air Fuel Ratio (Commanded)                       | float        | NULL    |         0 |    0 |    30 |
| kff1249         | Air Fuel Ratio (Measured)                        | float        | NULL    |         0 | NULL |  NULL |
| k12             | Air Status                                       | float        | NULL    |         0 | NULL |  NULL |
| k46             | Ambient Air Temp                                 | float        | &deg;C  |         1 |  -40 |    50 |
| kff1263         | Average Trip Speed (Whilst Moving Only)          | float        | km/h    |         1 |    0 |    16 |
| kff1272         | Average Trip Speed (Whilst Stopped or Moving)    | float        | km/h    |         1 |    0 |    16 |
| kff1270         | Barometer (On Android device)                    | float        | mb      |         0 |  800 |  1100 |
| k33             | Barometric Pressure (From Vehicle)               | float        | kPa     |         0 |    0 |   255 |
| k3c             | Catalyst Temperature (Bank 1 Sensor 1)           | float        | &deg;C  |         1 |    0 |    60 |
[... cut ...]
| profileWeight   | profileWeight                                    | float        | NULL    |         1 | NULL |  NULL |
| profileFuelCost | profileFuelCost                                  | float        | NULL    |         1 | NULL |  NULL |
| profileFuelType | profileFuelType                                  | varchar(255) | NULL    |         1 | NULL |  NULL |
+-----------------+--------------------------------------------------+--------------+---------+-----------+------+-------+
151 rows in set (0.06 sec)

```

aldodemi, 2016-07-11:
In my raw_logs table I have my old trip that doen't have a corresponding record in the sessions table so I can not see it in the web gui.
So all I need I think, in my case, is how to create sessions entry from my old trip record.

surfrock66, 2016-07-11:
Ok, good.  Even better.

A lot of the profile information isn't recorded very well, so let's call those fields optional.  To be honest, they show up 50% of the time, and I haven't figured out why...every so often I just update all entries in the session table.  To that end, let's consider the following values static as long as you're using the same profile:

eml
profileName
profileFuelType
profileWeight
profileVe
profileFuelCost

Run this query:

`use torque; select distinct session, id, v, min(time) as time, min(time) as timestart, max(time) as timeend, count(time) as sessionsize from raw_logs group by session, id, v;`

I took the results of that, dumped it into libreoffice calc, then did a concatenate on every row to create  an insert string that also put in the static values.  Something like "CONCATENATE("insert into sessions (v, id, session, time, eml, profileName, timestart, timeend, sessionsize, profileFuelType, profileWeight, profileVe, profileFuelCost) VALUES(",A2,",",B2,",",C2,",",D2,",'aldodemi@domain.com','CoolCarProfile',",E2,",",F2,",",G2,",'4','2000','85','.6')")" then dragged it down.

Thats an example, substitute your own vals of course.  I'm sure there is a more graceful way to that avoiding the work in office, but it'll work.

closed 2017-01-15 by surfrock66