surfrock66/torque #7: MySQL Syntax Error

marvinwankersteen, 2016-09-24:
Hey,

I like your Torque Viewer. I use an old version and it works well.
Now, I would like to upgrade the version and I get MySQL Syntax Errors. I think I have the wrong MySQL-Version installed.

The first error comes at the import of the three .sql-files:
```
# mysql -p < create_torque_keys_table.sql 
ERROR 1064 (42000) at line 3: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near '[DATABASE_NAME]
    AND table_type = 'BASE TABLE'
    AND table_name = 'torque_k' at line 4
```

The second errors when I open the /session.php:
`You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'WHERE ( YEAR(FROM_UNIXTIME(session/1000)) LIKE '2016' AND MONTHNAME(FROM_UNIXTI' at line 1`

Which version from MySQL should I use? I have installed "5.5.52-0ubuntu0.14.04.1".

It would be nice if you can help me =)

Marvin

surfrock66, 2016-09-24:
Curious...I'm on "5.7.15-0ubuntu0.16.04.1" and have no errors in my log.  Still, there shouldn't be anything in there that is version specific...lemme take a look in the morning.

surfrock66, 2016-09-26:
So I had built some code to not screw over people that were migrating over from an old torque install; basically if it found "raw_logs" it needs to move it to "raw_logs_old".  It looks like that doesn't work.  Lemme try to fix that.

surfrock66, 2016-09-26:
Fixed, try again, thanks for catching that!

closed 2016-09-26 by surfrock66

marvinwankersteen, 2016-09-27:
Nice, thanks!
I will try it and give you feedback.

marvinwankersteen, 2016-09-27:
It works! Thanks!