surfrock66/torque #10: No sessions being displayed

Crasha, 2016-10-01:
The data is there and I could see all my sessions previously. Checked just now and it's not displaying any sessions even though the data exists in the database.

surfrock66, 2016-10-03:
Do you have anything in your mysql logs?  That error indicates it couldn't fetch any sessions; since the data is there I'm wondering if something isn't wonky with the retrieval of sessions in the get_sessions.php file.  I haven't changed anything in forever, but it's possible if malformed data got in there it could mess up the query, or it could be an input-santiization bug.

Crasha, 2016-10-03:
Hmm, I see one log from yesterday when I drove home, but none of my other logs.

Here's all I see in the SQL logs;

> 2008 Init DB   TorqueLog
>                  2008 Query     SELECT timestart, timeend, session, profileName, sessionsize FROM sessions WHERE ( YEAR(FROM_UNIXTIME(session/1000)) LIKE '2016'  AND MONTHNAME(FROM_UNIXTIME(session/1000)) LIKE 'October'  )  GROUP BY session, profileName, timestart, timeend, sessionsize ORDER BY session DESC
>                  2008 Query     SELECT id,description,type FROM torque_keys WHERE populated = 1 ORDER BY description
>                  2008 Query     SELECT kff1006, kff1005 FROM raw_logs
>               WHERE session=1475427371343
>               ORDER BY time DESC
>                  2008 Query     SELECT YEAR(FROM_UNIXTIME(session/1000)) as 'year'
>               FROM sessions WHERE session <> ''
>               GROUP BY YEAR(FROM_UNIXTIME(session/1000))
>               ORDER BY YEAR(FROM_UNIXTIME(session/1000))
>                  2008 Query     SELECT distinct profileName FROM sessions ORDER BY profileName asc
>                  2008 Quit

surfrock66, 2016-10-03:
UGGGHHHHH I forgot about this and thought I solved it, and it's been so long since looking at the code I forgot.  At the first of the month the default filter applies to an empty set (it says "check October of this year for sessions") and that stops subsequent queries...you added an October session, now the script continues.  I thought I patched that by doing a "pull the last 2 months" code...I'll review.

Crasha, 2016-10-04:
Heheh, glad you remembered! Will you let me know when I can update from the repo with the fixes?

surfrock66, 2016-10-05:
Which version are you using?  I was looking back through my commits, see this issue:

https://github.com/surfrock66/torque/issues/1

Then this commit to solve it:

https://github.com/surfrock66/torque/commit/66bcd4dcf784e80f88f76ae7c356d6ae8aeac836

I'm reviewing the code, and it looks fixed.  Make sure you're up to date?

marvinwankersteen, 2017-01-02:
I have a similar problem. Since January 01 2017 no session are being displayed. I got no error or something else. I also cant select the year 2016. If I look in the database, there are the entries from 2016. I use the code from the last commit 9e61be8.

surfrock66, 2017-01-05:
I need to take another look at the condition where it deals with finding an empty set.  If I can get some code time tonight while I'm looking at https://github.com/surfrock66/torque/issues/12 I'll see if I can handle it better.  I still think the core issue is the rolling over of the default set when a new month comes around...I may have solved the "new month" condition, but not the "new year" condition.

adel-s, 2017-01-08:
The same error with current commit - no sessions. In mysql log I see only two requests:

2017-01-08T17:35:15.829271Z  240 Query	SELECT timestart, timeend, session, profileName, sessionsize FROM sessions WHERE ( YEAR(FROM_UNIXTIME(session/1000)) LIKE '2017'  AND MONTHNAME(FROM_UNIXTIME(session/1000)) LIKE 'January'  )  GROUP BY session, profileName, timestart, timeend, sessionsize ORDER BY session DESC

2017-01-08T17:35:15.832585Z  240 Query	SELECT id,description,type FROM torque_keys WHERE populated = 1 ORDER BY description

When I repeat first query in MYSQL I get 8 sessions. Second query returns torque keys.

And in webserver logs I see the same error as mentioned above:
PHP message: PHP Warning:  max(): When only one parameter is given, it must be an array

marvinwankersteen, 2017-01-10:
It is working now. My problem was, that there were no logs sent to the server. The bluetooth-adapter had no connection to the ECU. Replugged in and all fine.

Now, I have the first logs in 2017 and the session are displayed.

Sorry for my english.

surfrock66, 2017-01-10:
@marvinwankersteen  glad to hear it!

@adel-s Do me a favor, in the mysql console, run the following (correct your DB name of course)
`use torque;
SELECT session, YEAR(FROM_UNIXTIME(session/1000)) as year, MONTHNAME(FROM_UNIXTIME(session/1000)) as month from raw_logs ORDER BY session DESC LIMIT 20;
SELECT session, YEAR(FROM_UNIXTIME(session/1000)) as year, MONTHNAME(FROM_UNIXTIME(session/1000)) as month from sessions ORDER BY session DESC LIMIT 20;`

Then paste the results please.

adel-s, 2017-01-10:
mysql> SELECT session, YEAR(FROM_UNIXTIME(session/1000)) as year, MONTHNAME(FROM_UNIXTIME(session/1000)) as month from raw_logs ORDER BY session DESC LIMIT 20;
+---------------+------+---------+
| session       | year | month   |
+---------------+------+---------+
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887644975 | 2017 | January |
| 1483887339369 | 2017 | January |
| 1483887339369 | 2017 | January |
| 1483887339369 | 2017 | January |
| 1483887339369 | 2017 | January |
| 1483887339369 | 2017 | January |
| 1483887339369 | 2017 | January |
| 1483887339369 | 2017 | January |
+---------------+------+---------+
20 rows in set (0,00 sec)

mysql> SELECT session, YEAR(FROM_UNIXTIME(session/1000)) as year, MONTHNAME(FROM_UNIXTIME(session/1000)) as month from sessions ORDER BY session DESC LIMIT 20;
+---------------+------+---------+
| session       | year | month   |
+---------------+------+---------+
| 1483887644975 | 2017 | January |
| 1483887339369 | 2017 | January |
| 1483887074449 | 2017 | January |
| 1483887020227 | 2017 | January |
| 1483886904548 | 2017 | January |
| 1483886821858 | 2017 | January |
| 1483886719574 | 2017 | January |
| 1483886622060 | 2017 | January |
+---------------+------+---------+
8 rows in set (0,00 sec)

But after some debug today I found why this happened - all my sessions has $row["sessionsize"] less than 20. That's why I see no sessions.

Sorry my bad.

surfrock66, 2017-01-10:
Oh, ok cool, glad it's working!

closed 2017-01-10 by surfrock66