surfrock66/torque #11: Pre-select favorite variables

marvinwankersteen, 2016-10-12:
Its not really an issue, but its a bit disturbing that the selected variables are not saved if I choose a other session.
Is it possible to pre-select my favorite variables? I have tried to hardcode it in the code, but its hard for me to unterstand =)

surfrock66, 2016-10-16:
Hmm, it'd be tricky.  Not impossible, but I don't know where you'd inject the routine:

First, when the list of variables is populated, it reads from the torque.keys table.  You could add a binary variable that indicates whether it's pre-selected or not.  The trick is, on submit, you'd need to then update the keys table with any changes...that's a bunch of extra I/O, not impossible, but not streamlined.        I'd have to look into it.

surfrock66, 2019-02-03:
Ok 2 years later...this is now done, I'll be pushing a major code revision and it'll have this in it.  It doesn't auto-plot, but you can select favorite variables then plotting is 1 button away.

closed 2019-02-03 by surfrock66