surfrock66/torque #12: Use mysqli instead of mysql

smasher816, 2017-01-04:
```
Fatal error: Uncaught Error: Call to undefined function mysql_connect() in /srv/http/torque/db.php:7 Stack trace: #0 /srv/http/torque/session.php(7): require_once() #1 {main} thrown in /srv/http/torque/db.php on line 7
```

http://php.net/manual/en/migration55.deprecated.php

surfrock66, 2017-01-05:
Huh, what's your OS and php level?  Mine is still working, but it seems there's a need to migrate...I don't know when I'm gonna get to finishing this as my coding time is 0 right now, if you want to patch it I'll certainly accept it.

surfrock66, 2017-01-08:
OK, I think this is fixed in these 2 commits:

https://github.com/surfrock66/torque/commit/05f76edce95bec5f3331a3913ccb10c406f3b47f
https://github.com/surfrock66/torque/commit/44bc4606d01c69faaf474c7816268776773eb056

It was a big pain in the ass...the i variants require a different order of arguments for no reason.  I scrubbed everything I could find.  Please help test.

closed 2017-01-08 by surfrock66