surfrock66/torque #14: Can not switch between sessions.

adel-s, 2017-01-14:
- How to reproduce:
-- Open man window (session.php)
-- At sessions dropdown menu check any session
-- Got url: /torque/session.php?id= and error message "You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'ORDER BY time DESC' at line 3"

- What happened
-- After selecting session we got 'OnChange' event, submit form and call url.php with POST request with variable 'seshidtag' = selected session
-- url.php got an error on line 12
`$seshid = strval(mysqli_escape_string($_POST["seshidtag"]));`
because `mysqli_escape_string` function requres mysql link as first parameter.
-- variable $seshid becomes unset
-- session.php got empty session id and tries to make mysql query without session id.

- How to fix:
I see two options:
1. Simple change mysqli_escape_string to mysql_escape_string in url.php on lines 10 and 12
2. Provide mysql link to function. As I see when we call url.php we have no connection to mysql yet.

For myself I fixed it by 1st option for now, but maybe you see more correct solution.

surfrock66, 2017-01-15:
The crappy thing is, this whole mysqli thing is because the mysql_ syntax isn't just depreciated...it's being removed.  It has to move.  Lemme test a fix.

surfrock66, 2017-01-15:
Commit https://github.com/surfrock66/torque/commit/219a42d66b32dc63c46e65ba09311da3edc37fd1 has it fixed for me, see if it's fixed for you.

adel-s, 2017-01-15:
Confirm. It works like a charm. Thanks!

closed 2017-01-15 by surfrock66