surfrock66/torque #19: raw_logs are populated, nothing in sessions

dscheste, 2017-02-24:
Torque seems to log data properly, the raw_logs table is populated with a bunch of logs, but the sessions table remains empty. Nothing there at all. Empty.

I see the following in the /var/log/apache2/other_vhosts_access.log:
```
127.0.0.1:80 199.119.233.195 - - [23/Feb/2017:09:16:38 -0500] "GET /torque/upload_data.php?eml=dscheste@gmail.com&v=8&session=1487859162086&id=4cf42e112c09ab851496fd59099ffbc5&time=1487859393294&kff1005=-79.52420459&kff1006=43.72671025&kff1001=41.581264&kff1007=170.6&kff124d=14.289523&kff1249=15.266144&k46=15.0&k5=88.0&k4=16.862745&kc=1478.0&k2f=29.80392&kff1010=105.0&kff123b=170.6&kff1006=43.72671025&kff1005=-79.52420459&kf=16.0&kb=56.0&kd=41.0&kfe1805=89.0 HTTP/1.0" 200 169 "-" "-"
127.0.0.1:80 199.119.233.195 - - [23/Feb/2017:09:17:35 -0500] "GET /torque/upload_data.php?eml=dscheste@gmail.com&v=8&session=1487859162086&id=4cf42e112c09ab851496fd59099ffbc5&time=1487859453295&kff1005=-79.52220189&kff1006=43.72075943&kff1001=45.04296&kff1007=152.9&kff124d=14.423209&kff1249=18.751831&k46=15.0&k5=87.0&k4=16.470589&kc=1661.25&k2f=29.80392&kff1010=97.0&kff123b=152.9&kff1006=43.72075943&kff1005=-79.52220189&kf=16.0&kb=25.0&kd=47.0&kfe1805=86.0 HTTP/1.0" 200 169 "-" "-"
127.0.0.1:80 199.119.233.195 - - [23/Feb/2017:09:18:35 -0500] "GET /torque/upload_data.php?eml=dscheste@gmail.com&v=8&session=1487859162086&id=4cf42e112c09ab851496fd59099ffbc5&time=1487859513294&kff1005=-79.52777046&kff1006=43.71553919&kff1001=97.40557&kff1007=252.2&kff124d=14.721085&kff1249=14.759665&k46=15.0&k5=88.0&kff1273=39.55947&k4=96.07843&kc=2605.75&k2f=34.901962&kff1010=103.0&kff123b=252.2&kff1006=43.71553919&kff1005=-79.52777046&kf=16.0&kb=98.0&kd=101.0&kfe1805=88.0 HTTP/1.0" 200 169 "-" "-"

```

200 status code means the API call was successful, not sure what 169 means.

How do I troubleshoot it from here?

My understanding is that this GET call should have inserted values into the sessions table, but evidently, it has not.

Thank you.

dscheste, 2017-02-27:
I think I have gotten it to work by removing $sessprofilequery at the end of the sessions query in the upload_data.php:

` $sessionqrystring = "INSERT INTO $db_sessions_table (".quote_names($sesskeys).", timestart, sessionsize) VALUES (".quote_values($sessvalues).", $sesstime, '1') ON DUPLICATE KEY UPDATE timeend='$sesstime', sessionsize='$sesssizecount'**$sessprofilequery**";
//echo $sessionqrystring;
`

As a side note, why don't you use a trigger on the mysql side instead of running two queries in php?

closed 2017-02-27 by dscheste

surfrock66, 2017-02-27:
Weird, I'm not having this issue at all.

I'd accept a patch for the trigger thing; I don't know mysql deep enough to be able to implement that.