surfrock66/torque #22: Unable to load data from app

aldodemi, 2018-01-09:
Server Debian 9/MySql 5.7.20, unable to load data anymore from the app to the database.
I cannot see any error in apache2 error.log, a lot of record in access.log.
Example: 9Y.6Z.1HH.1KK - - [09/Jan/2018:16:46:07 +0100] "GET /upload_data.php?eml=aldo.demicheli@MYADDRESS.IT&v=8&session=1515509174707&id=5d70f4589df3a0ceaf3ae9d4609ddb5f&time=1515509303128&kff1005=8.65746931&kff1006=44.40322219&kff1001=4.284&kff1007=62.5&kff5202=10.190329&kff5203=9.81314&kff1204=29.456383&kff120c=16640.154&kff1239=23.0&kff1010=80.0&kff1006=44.40322219&kff1005=8.65746931&kff123b=62.5&kff123a=6.0&kff1296=32.57341&kff1297=0.0&kff1298=67.42659&kff1270=1011.55566&kff1226=0.10033698&kff1273=0.07482127&kff1220=-0.02166008&kff1221=2.4405724E-4&kff1222=0.9672623&kff1223=0.06784683&kff1267=62.022&kff1266=91.0&kff1268=29.962&kff1001=4.284 HTTP/1.0" 200 214 "-" "-"
Closing the app it stay in "Uploading logs"... No data into the database

surfrock66, 2018-01-10:
Do you have mysql logs?  The data is getting to the web server, but the web server isn't communicating with the database, or at least isn't writing.  If you load the torque viewer, do you get any apache errors?

aldodemi, 2018-01-10:
No mysql errors in the mysql logs, if I try to connect to “upload_data.php” via web I receive the “regular” OK message, the torque viewer does work but it's empty…

surfrock66, 2018-01-10:
Have you ever been able to upload logs, or is it a new issue?  There is some sort of miscommunication between the database and the website; because of that I'm surprised we're not seeing any issues in the apache access logs, the apache error logs, or the mysql logs.  

Just to be sure, can you validate the mysql username, mysql password, mysql database names, mysql table names, and access permissions for that account to the DB/Table?  Without an error to investigate, I'm not quite sure where to start.

aldodemi, 2018-01-10:
You are right, sorry, lost to give some information...
This evening I do some other test and, with Debian 8.1 and mysql 5.5.58, all works fine (on another pc) don't know if the problem, than, is Debian or, as I think, mysql version...

aldodemi, 2018-01-10:
Downgraded mysql server on Debian 9 to 5.6 still does't works...

surfrock66, 2018-01-11:
Odd.  I'm not really sure how to troubleshoot with no errors in the logs.  

 You can simulate some data upload by just loading the following URL in a web browser:
`http://IP/torque/upload_data.php?eml=aldo.demicheli@MYADDRESS.IT&v=8&session=1515509174707&id=5d70f4589df3a0ceaf3ae9d4609ddb5f&time=1515509303128`

Just iterate the time and you can create all the dummy data you want.

If no mysql entries or errors show up there, we'll have to think of a different troubleshooting test.

aldodemi, 2018-01-11:
Test result:
Field 'timeend' doesn't have a default value

aldodemi, 2018-01-11:
Create a new Debian box from scratch this morning with debian 8.1 and mysql server version: 5.5.58-0, all work as expected...

surfrock66, 2018-01-11:
Hmm, so it's working on that other box?  I run mine on Ubuntu 17.10 with mysql 5.7.20-0ubuntu0.17.10.1 and Apache 2.4.27.  This has been running since much older apache and mysql versions...continuous upgrades since probably 2015.

aldodemi, 2018-01-12:
Recreate this morning the other box from scrath with Ubuntu 16.04.3 LTS (17.10 is no more available)  with mysql 5.7.20-0ubuntu0.16.04.1: no way!!!
Torque app seem not load data but... data are loaded, but only in row_logs table, sessions table is empty 
[image]
No error with manual load test in debian 8/mysql 5.5, same error (Field 'timeend' doesn't have a default value) in ubuntu 16/mysql 5.7

aldodemi, 2018-01-13:
Reinstalled form scratch two different server, both with Debian 8.1 and MySQL 5.5, both work fine....

surfrock66, 2018-01-13:
huh.  So, what were the configs that didn't work again?

aldodemi, 2018-01-15:
using debian 8 and mysql 5.5 works, using debian 9 and mysql 5.6 or 5.7 does not works.

SolverTwins, 2018-08-29:
I have the same problem, and I report you how I solve.
I use Windows 2012 server, and MySQL 5.7, so I know thay I will have problems, but I'll try always to solve it.
The problem with the versions of MySQL and the APP, its that the STRICT mode its enable by default in 5.7 version. To solve it.
First, check whether the strict mode is enabled or not in mysql using:
SHOW VARIABLES LIKE 'sql_mode'; If you want to disable it:
SET sql_mode = ''; or any other mode can be set except the following. To enable strict mode:
SET sql_mode = 'STRICT_TRANS_TABLES'; You can check the result from the first mysql query.

Well, it is the one method, I investigate again and I found that the problems are with the NULL values of several fields in the database, so I permit the null values in the timeend field of the sessions database, and it also works.

CompuStation, 2019-03-12:
Using Ubuntu 14.04 and 
mysql  Ver 14.14 Distrib 5.5.62 
PHP 5.5.9-1ubuntu4.27 works fine.
Latest Ubuntu  and mysql 5.6 or 5.7 does not works.

surfrock66, 2019-03-12:
Do you have anything in your apache, or mysql logs?  Usually "/var/log/apache2/error.log" and "/var/log/mysql/error.log".  I would bet something would show up in the former.

Does the website load at all?  

My testing is on Ubuntu 18.10, php 7.2.15, mysql 14.14, apache 2.4.29.

LeoRX, 2019-05-27:
So glad I found this.
my sql_mode was set to
STRICT_TRANS_TABLES,​ERROR_FOR_DIVISION_BY_ZERO,​NO_AUTO_CREATE_USER,​NO_ENGINE_SUBSTITUTION

removing it made it work :)  Thanks @SolverTwins

MickLC, 2019-07-24:
My problem was the same (Field 'timeend' doesn't have a default value).

My solution was to alter sessions:

ALTER TABLE sessions MODIFY timeend varchar(15) DEFAULT '';

My queue of log entries appears to be draining now.

surfrock66, 2019-07-24:
Glad it works; I worry about having a default in there, but it seems it is needed.  I committed a fix.

closed 2019-07-24 by surfrock66