surfrock66/torque #24: Polylines not matching road

nicolasraj, 2018-01-31:
I realized that the kff1005 | GPS Longitude column only has at most 3 decimal places and this this making the path jagged/irregular. Is there any fix for this ?

surfrock66, 2018-02-02:
Honestly yes and there's a patch in the queue that fixes this.  I haven't released it yet because I'm doing a major codebase redesign.  Take a look here; you can do this right now: https://github.com/surfrock66/torque/pull/21

I want to add some logic to make sure there's nothing breaking before just pushing that.  My codebase rewrite will address a few things:

1) Try to add logic to prevent the following issue AND to clear data if this gets in: https://github.com/surfrock66/torque/issues/23
2) Purge data duplicated between the session and the raw_logs table
3) Come up with logic to split raw_logs into tables by year (raw_logs_2018) to speed up queries upon big datasets.

surfrock66, 2019-01-26:
This is fixed in my local repo, but requires major DB modification which I'm ironing out.  It'll be released in my next update hopefully within the week.

closed 2019-01-26 by surfrock66