surfrock66/torque #25: Temperature

5ft24dave, 2018-03-24:
Maybe I'm not seeing it, but is there an option to display the temperature data in Degrees F instead of degrees C?
Thanks

surfrock66, 2018-03-25:
I see you closed issue 51 on the root repository, did you figure it out?

5ft24dave, 2018-03-25:
No, thought that one was for the torque app itself.  Have that one set to Deg F.  Just not the web server.  It shows Deg C and haven't been able to find it

⁣Sent from BlueMail ​

surfrock66, 2018-03-27:
It's a setting you set in "creds.php" in the viewer's root web directory.  Something like this (should be the same, but this is from my development branch so I forget if there's a difference).  Look for "source is F" and "Use F":

`surfrock66@sr66-blade(shell):torquetest$ cat creds.php 
<?php
//echo "\r\n";

// MySQL Credentials
$db_host = 'localhost';
$db_user = 'REDACTED';     // Enter your MySQL username
$db_pass = 'REDACTED';     // Enter your MySQL password
$db_name = 'REDACTED';
$db_table = 'raw_logs';
$db_keys_table = 'torque_keys';
$db_sessions_table = 'sessions';
$gmapsApiKey = 'REDACTED'; // OPTIONAL: Create a key at https://developers.google.com/maps/documentation/javascript/

// User credentials for Browser login
$auth_user = '';    //Sample: 'torque'
$auth_pass = '';    //Sample: 'open'

//If you want to restrict access to upload_data.php, 
// either enter your torque ID as shown in the torque app, 
// or enter the hashed ID as it can found in the uploaded data.
//The hash is simply MD5(ID).
//Leave empty to allow any torque app to upload data to this server.
$torque_id = 'REDACTED';        //Sample: 123456789012345
$torque_id_hash = 'REDACTED';   //Sample: 58b9b9268acaef64ac6a80b0543357e6
//Just 'settings', could be moved to a config file later.
$source_is_fahrenheit = false;
$use_fahrenheit = true;

$source_is_miles = false;
$use_miles = true;

//$hide_empty_variables = true;
$hide_empty_variables = false;
$show_session_length = true;
//echo "\r\n";

?>
`

5ft24dave, 2018-03-27:
> maps/documentation/javascript/ // User credentials for Browser login

Cool,
Thank you

5ft24dave, 2018-03-28:
Cool, works perfect.  Thanknyou

⁣Sent from BlueMail ​

closed 2018-03-28 by surfrock66