surfrock66/torque #26: Multiple New Features Planned

surfrock66, 2019-02-04:
From forks, I want to implement the following features and fixes:

1. Rely on "empty" and not !isset: https://github.com/aegixx/torque/commit/81a43a52a1356d07267f253883bdc2070a096369
2. Script for processing the upload of a session from a csv export in the torque app: https://github.com/adel-s/torque/commit/a1b3eb89a03807a79bd3d7ca789a634c7a0903db
3. Make the minimum session size configurable: https://github.com/adel-s/torque/commit/a0e6ff8b3fbab8f784a29071374f830b9adbbf08
4. Show events as map markers: https://github.com/adel-s/torque/commit/0f7113c7265c75efe9c560e38dda6f243feb0481
5. Evaluate redundant result freeing: https://github.com/adel-s/torque/commit/825c59d28b6223a28abac72f460ae0834b85aae2
6. Enable multiple user accounts: https://github.com/adel-s/torque/commit/bc5ece5619a29554a82f0c4d0e5f53814ec72284

stealer0517, 2019-02-19:
Not sure if this is the preferred way to add some feature requests, but I have two ideas that would be neat (and I'm far too dumb to figure out php).
1. Ability to zoom on the chart, or set min/maxes to display on the chart.
My mpg shoots up to 512 when I let go of the gas which makes it pretty much unusable.
I figured out how to make the chart taller which helps a lot.

2. Multiple charts.
Say I want to view Engine temperature, and the RPM the difference between 30-180(f) and 800-3000 is huge.
I'm sure this could be achieved with some copy pasting, but I'm seemingly too dumb to figure out the chart.

surfrock66, 2019-02-27:
#6 in my list is done via this commit: https://github.com/surfrock66/torque/commit/45e0747c6a8a0c505a0bd43b4fae71bd3da66f3e

surfrock66, 2019-03-11:
3 is done via commit https://github.com/surfrock66/torque/commit/cb06712d72b265b836a19582ede7e49c390d0f8a
5 is done via commit https://github.com/surfrock66/torque/commit/c09c64adc158e281b3fb12b5d06a0abbe266d12e

surfrock66, 2019-03-11:
I've done some research for 1 above, and I've decided to not implement this change; I may come back to it but for now the logic as I'm reading through works as is.  Here's some research: https://www.virendrachandak.com/techtalk/php-isset-vs-empty-vs-is_null/