surfrock66/torque #27: Trip distance in KM not MI

stealer0517, 2019-02-27:
In my creds.php file I have $use_miles = true;
But when I plot the trip distance it shows is: Trip Distance (km) and the value is in km instead of miles.
Other things are set to miles like speed in OBD and GPS. 
Just the trip distance seems to be rebelling.

surfrock66, 2019-02-27:
Well crap, this is kind of a major bug.  The code which does the plotting literally looks at the description string for the PID and searches for "Temp" or "Speed" when deciding to invoke the conversion...this doesn't make sense.  Distance doesn't work, things like mpg don't work...etc.  I probably need to rethink how that's done across the board to fix it; A dirty patch would be to add a search for the word distance (based on my PID list here that should cover it: https://hda.surfrock66.com/torquetest/pid_edit.php ) but it feels...incomplete.  I'll try to toss something together.

surfrock66, 2019-02-27:
Ok, fixed in this commit just like I said...it now filters on Distance.  If an edge case pops up maybe a better solution will present itself: https://github.com/surfrock66/torque/commit/1ad47c64bc109a0f7ef80eb0c9ea73b57268fcf9

closed 2019-02-27 by surfrock66