surfrock66/torque #29: Not an Issue, Just a Thanks

econpy, 2019-04-07:
I'm the original creator of this repo. Built from scratch while wasting time at my old academic job. I am in a different line of business now and just don't have the time to go back into it. However, I wanted to let you know surfrock66 that I get all the emails still about your changes and want to thank you for your continued hard work on this. So many hours went into it, a true labor of love. So thank you kindly for your continued hard work on this, I greatly appreciate it.
--Matt

surfrock66, 2019-04-08:
Thanks, I really appreciate it.  It's been a fun project for me to work on my git skills...I now have a 2 and 4 year old so my non-work coding time is less than I'd like it to be, but I have ideas from the subsequent forks and really love working on it.

deepak-gunasekaran, 2020-07-30:
Looks like the recent update killed the own server configuration. I could not configure my own server anymore. Is there any other workaround to bring back the configure your own server feature outside or moving back to older version of apk?