surfrock66/torque #33: db_port Undefined Variable error Apache logs

stealer0517, 2019-08-29:
I am getting the following error in my Apache logs:

`[Thu Aug 29 12:52:31.812383 2019] [php7:notice] [pid 22691] [client (IP):52533] PHP Notice:  Undefined variable: db_port in /var/www/html/torque/db.php on line 7, referer: http://(IP)/session.php`

It appears that in the creds.php file there wasn't a line for the db port so I added the following line and I stopped getting that error.

surfrock66, 2019-08-29:
This was from this patch: https://github.com/surfrock66/torque/commit/1a0e049c39691e46c55070e0c3e2755e02582882

You may wanna consider re-pulling as there's been a couple bug-fixes from there.

stealer0517, 2019-08-29:
I thought I downloaded it last night, but I'll try again when I get home.
I've been having a few other issues so I'll see if that fixes them as well.

surfrock66, 2019-08-29:
Ok, the big bug is in upload_data.php, look at the most recent commit (it's like 2 lines, so if you don't have it it's easy to patch).  New PIDs weren't getting auto-created.

stealer0517, 2019-08-30:
I tried starting over from scratch and I still got the error until I specified the port.