surfrock66/torque #35: New torque updated killed the remote server upload to any other webserver outside of theirs.

deepak-gunasekaran, 2020-07-30:
Even when I configure my own server it does not send the requests there. Looks like the recent app update removed this cool feature. Lemme know if you guys have this feature still working on your device or alternate option to make this work.

surfrock66, 2020-07-30:
I'm not sure what's going on for you, I actually use torque pro on Android which still has the "your own server" option (and the last update was 11/14/2019 to v1.10.114) but Torque Lite is version 1.2.22 and was last updated on 1/13/2018.  Can you give me more info about what changed for you?

deepak-gunasekaran, 2020-07-30:
In the latest version of torque pro, when I configure my own server in remote upload, I do not get even a single request coming to my webserver. I dug further into Charles proxy and could see that the app does not make any call to the remote server I configured in the app. Rather it always makes call to one of the google services to upload the data.

surfrock66, 2020-07-30:
Can you confirm the version of torque pro you're on?

deepak-gunasekaran, 2020-07-30:
1.10.114 is the version I'm on and facing this issue.

surfrock66, 2020-08-01:
I have the same version as you and I was able to confirm that I still have that setting and was able to upload a track successfully yesterday. I'm not sure what else to tell you, I am not affiliated with the torque app at all but probably you should submit a request to the developer.  

[image]

[image]

[image]

deepak-gunasekaran, 2020-08-01:
@surfrock66 - Thanks for taking the time and explaining this in detail. I tried this from K8 Istio environment and it failed all time. With setting this up on raspberry pi I realized the app makes HTTP/1.0 which is by default not supported by the istio :(. Took so long to figure this out. Anyway, this is not an issue with the tool but with the Istio gateway.

closed 2020-08-01 by deepak-gunasekaran