surfrock66/torque #36: New installation in new Ubuntu version (20.10 but also in previuos versions)

aldodemi, 2020-11-26:
Your MySQL connection id is 3772
Server version: 8.0.22-0ubuntu0.20.10.2 (Ubuntu)

If I try to upload from app or test via web (http:/mypublic.server:mypublicport//upload_data.php loading data doesn't works:

Via web server I have this error:

“You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near '' at line 1”

Via app no data loading...

This is a blank new server, the old one, with Debian 8, that worked is definitively out of order...

surfrock66, 2020-11-27:
Ok curious; I have had my version on my server updating since like 2014 or something so I haven't done a new setup in forever.  Is there anything in the mysql error log to help me debug?

aldodemi, 2020-12-01:
Sorry for the delay in my answer, here my sql log:

2020-11-30T12:07:12.763599Z 0 [System] [MY-013172] [Server] Received SHUTDOWN from user <via user signal>. Shutting down mysqld (Version: 8.0.22-0ubuntu0.20.10.2).
2020-11-30T12:07:13.537205Z 0 [System] [MY-010910] [Server] /usr/sbin/mysqld: Shutdown complete (mysqld 8.0.22-0ubuntu0.20.10.2)  (Ubuntu).
2020-11-30T12:09:53.145842Z 0 [System] [MY-010116] [Server] /usr/sbin/mysqld (mysqld 8.0.22-0ubuntu0.20.10.2) starting as process 965
2020-11-30T12:09:53.162917Z 1 [System] [MY-013576] [InnoDB] InnoDB initialization has started.
2020-11-30T12:09:53.947060Z 1 [System] [MY-013577] [InnoDB] InnoDB initialization has ended.
2020-11-30T12:09:54.124216Z 0 [System] [MY-011323] [Server] X Plugin ready for connections. Bind-address: '127.0.0.1' port: 33060, socket: /var/run/mysqld/mysqlx.sock
2020-11-30T12:09:54.226233Z 0 [Warning] [MY-010068] [Server] CA certificate ca.pem is self signed.
2020-11-30T12:09:54.226462Z 0 [System] [MY-013602] [Server] Channel mysql_main configured to support TLS. Encrypted connections are now supported for this channel.
2020-11-30T12:09:54.257564Z 0 [System] [MY-010931] [Server] /usr/sbin/mysqld: ready for connections. Version: '8.0.22-0ubuntu0.20.10.2'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  (Ubuntu).

No errors are logged when try to access by "upload_data.php"

aldodemi, 2020-12-01:
In the past my server was a Debian 8, torque worked, when crashed (SSD fault) I recreated it with Debian 10 but the same error appared so I recreated with Debian 8 and all started to work again. Now that this old server is full out of order I try to install Torque on an existring Ububtu 20.10 server

aldodemi, 2020-12-01:
session.php works fine (in this moment the dabase is empty but if I load via restore some data those are normaly shows)
[image]

aldodemi, 2020-12-02:
This is, for example, my sql log today after had try to connect via web and via app:
2020-12-02T07:19:49.800296Z 0 [System] [MY-010116] [Server] /usr/sbin/mysqld (mysqld 8.0.22-0ubuntu0.20.10.2) starting as process 941
2020-12-02T07:19:49.826165Z 1 [System] [MY-013576] [InnoDB] InnoDB initialization has started.
2020-12-02T07:19:50.589908Z 1 [System] [MY-013577] [InnoDB] InnoDB initialization has ended.
2020-12-02T07:19:50.802848Z 0 [System] [MY-011323] [Server] X Plugin ready for connections. Bind-address: '127.0.0.1' port: 33060, socket: /var/run/mysqld/mysqlx.sock
2020-12-02T07:19:50.840442Z 0 [System] [MY-010229] [Server] Starting XA crash recovery...
2020-12-02T07:19:50.852321Z 0 [System] [MY-010232] [Server] XA crash recovery finished.
2020-12-02T07:19:50.977971Z 0 [Warning] [MY-010068] [Server] CA certificate ca.pem is self signed.
2020-12-02T07:19:50.978344Z 0 [System] [MY-013602] [Server] Channel mysql_main configured to support TLS. Encrypted connections are now supported for this channel.
2020-12-02T07:19:51.008683Z 0 [System] [MY-010931] [Server] /usr/sbin/mysqld: ready for connections. Version: '8.0.22-0ubuntu0.20.10.2'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  (Ubuntu).

aldodemi, 2020-12-02:
Where can I see to try to find some more information?

aldodemi, 2020-12-02:
Can be in apache2.log somethings usefull?

[Wed Dec 02 16:08:37.657673 2020] [php7:notice] [pid 15830] [client 192.168.10.34:42816] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:37.659462 2020] [php7:notice] [pid 15558] [client 192.168.10.34:42818] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:38.674373 2020] [php7:notice] [pid 14333] [client 192.168.10.34:42824] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:38.681102 2020] [php7:notice] [pid 14840] [client 192.168.10.34:42828] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:38.681166 2020] [php7:notice] [pid 15449] [client 192.168.10.34:42826] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:38.685244 2020] [php7:notice] [pid 14536] [client 192.168.10.34:42834] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:38.690884 2020] [php7:notice] [pid 14222] [client 192.168.10.34:42838] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:38.690937 2020] [php7:notice] [pid 15342] [client 192.168.10.34:42836] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:38.692503 2020] [php7:notice] [pid 15558] [client 192.168.10.34:42832] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:38.692651 2020] [php7:notice] [pid 15830] [client 192.168.10.34:42842] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:38.694666 2020] [php7:notice] [pid 14930] [client 192.168.10.34:42830] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:38.694804 2020] [php7:notice] [pid 13454] [client 192.168.10.34:42840] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:38.705953 2020] [php7:notice] [pid 14333] [client 192.168.10.34:42844] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:39.731667 2020] [php7:notice] [pid 14840] [client 192.168.10.34:42848] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:39.754456 2020] [php7:notice] [pid 14536] [client 192.168.10.34:42862] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:39.758957 2020] [php7:notice] [pid 15449] [client 192.168.10.34:42852] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:39.759401 2020] [php7:notice] [pid 15558] [client 192.168.10.34:42854] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:39.760258 2020] [php7:notice] [pid 15342] [client 192.168.10.34:42850] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:39.761670 2020] [php7:notice] [pid 14930] [client 192.168.10.34:42858] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:39.762067 2020] [php7:notice] [pid 14222] [client 192.168.10.34:42860] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:39.762666 2020] [php7:notice] [pid 13454] [client 192.168.10.34:42866] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:39.763051 2020] [php7:notice] [pid 15830] [client 192.168.10.34:42868] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:39.763710 2020] [php7:notice] [pid 14333] [client 192.168.10.34:42856] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:39.764715 2020] [php7:notice] [pid 14840] [client 192.168.10.34:42864] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:40.790291 2020] [php7:notice] [pid 15449] [client 192.168.10.34:42876] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:40.790314 2020] [php7:notice] [pid 15558] [client 192.168.10.34:42874] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:40.791044 2020] [php7:notice] [pid 14930] [client 192.168.10.34:42878] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:40.792510 2020] [php7:notice] [pid 14333] [client 192.168.10.34:42880] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:40.794212 2020] [php7:notice] [pid 13454] [client 192.168.10.34:42882] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:40.794523 2020] [php7:notice] [pid 14840] [client 192.168.10.34:42884] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:40.795301 2020] [php7:notice] [pid 15342] [client 192.168.10.34:42890] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[... cut ...]
[Wed Dec 02 16:08:40.796177 2020] [php7:notice] [pid 15830] [client 192.168.10.34:42888] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:08:40.804191 2020] [php7:notice] [pid 15449] [client 192.168.10.34:42892] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8
[Wed Dec 02 16:38:55.827192 2020] [php7:notice] [pid 14333] [client 192.168.10.35:55727] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8

aldodemi, 2020-12-02:
This is in my apache2 "access.log" (aldo@iiiiiiiiiii.it is a fake address :-))

192.168.10.34 - - [02/Dec/2020:16:08:40 +0100] "GET /upload_data.php?eml=aldo@iiiiiiiiiii.it&v=8&session=1606914493645&id=0f607264fc6318a92b9e13c65db7cd3c&time=1606914626516&kff1005=8.950457696363827&kff1006=44.40110186089179&kff1001=0.103648536&kff1007=358.01825&ke=17.5&k4=26.27451&k43=13.333333&k7=-1.5625&k6=-6.25&kff1237=0.103648536&kff120c=16.269817&k21=0.0&k31=28732.0&kff1269=50.0&k2c=0.0&kc=1221.75&kff1239=4.0&kff1010=51.39289927061617&kff1006=44.40110186089179&kff1005=8.950457696363827&kff123b=358.01825&kff123a=14.0&k24=1.0067444&k34=1.0057983&kff1215=0.74&kff1298=62.876686&kff1297=0.0&kff1296=37.123314&k11=16.470589&k47=48.627453&k45=0.0&kff1270=1008.59424&k33=99.0&kb=26.0&kff1202=-10.587755&kff124d=14.7&kff1249=14.799142&k44=1.0&kff1220=-0.01976924&kff1221=0.8842464&kff1222=0.3673839&kff1223=7.0680253E-4&k10=4.09&k3c=510.0&k3e=384.5&kf=12.0&k5=55.0&kff1267=82.647&kff1266=131.0&k1f=521.0&kff1268=48.796&kff1238=13.8&k42=14.628&kff1240=3.3341064&kff1001=0.103648536&kd=0.0&kff1272=2.2857144&kff1263=24.0 HTTP/1.0" 200 466 "-" "-"
192.168.10.34 - - [02/Dec/2020:16:08:40 +0100] "GET /upload_data.php?eml=aldo@iiiiiiiiiii.it&v=8&session=1606914493645&id=0f607264fc6318a92b9e13c65db7cd3c&time=1606915280516&kff1005=8.940340053899249&kff1006=44.39879249409961&kff1001=32.958675&kff1007=63.374832&ke=5.0&k4=0.0&k43=0.0&k7=0.78125&k6=0.0&kff1237=0.83096313&kff1204=1.9297625&kff120c=18.19958&k21=0.0&k31=28734.0&k2c=0.0&kc=0.0&kff1239=3.0&kff1010=75.28551734524049&kff1006=44.39879249409961&kff1005=8.940340053899249&kff123b=63.374832&kff123a=13.0&k24=1.0655518&k34=1.0718384&kff1215=0.85499996&kff1298=57.41482&kff1297=0.0&kff1296=42.58518&k11=16.078432&k47=48.235294&k45=0.0&kff1270=1005.1548&k33=99.0&kb=100.0&kff1202=0.14503765&kff124d=14.067463&kff1249=15.66361&k44=0.9569702&kff1220=-0.15864205&kff1221=0.8710669&kff1222=0.58069646&kff1223=0.11719513&k10=0.48&k3c=617.6&k3e=433.3&kf=16.0&k5=60.0&kff1267=399.962&kff1266=785.0&k1f=1165.0&kff1268=385.755&kff1238=13.7&k42=14.648&kff1240=3.5587158&kff1001=32.958675&kd=35.0&kff1272=11.515625&kff1263=23.396826 HTTP/1.0" 200 466 "-" "-"
192.168.10.34 - - [02/Dec/2020:16:08:40 +0100] "GET /upload_data.php?eml=aldo@iiiiiiiiiii.it&v=8&session=1606914493645&id=0f607264fc6318a92b9e13c65db7cd3c&time=1606914967517&kff1005=8.952032219899616&kff1006=44.40181626036443&kff1001=25.813221&kff1007=196.13051&ke=23.5&k4=25.882353&k43=18.82353&k7=-1.5625&k6=0.78125&kff1237=0.071704865&kff1204=0.3368622&kff120c=16.60668&k21=0.0&k31=28733.0&kff1269=52.0&k2c=0.0&kc=1305.25&kff1239=6.0&kff1010=51.32953008952615&kff1006=44.40181626036443&kff1005=8.952032219899616&kff123b=196.13051&kff123a=10.0&k24=1.0065308&k34=1.0435181&kff1215=0.79999995&kff1298=77.397934&kff1297=0.0&kff1296=22.602068&k11=19.215687&k47=50.19608&k45=1.5686275&kff1270=1008.49805&k33=99.0&kb=27.0&kff1202=-10.442717&kff124d=14.7&kff1249=14.796002&k44=1.0&kff1220=0.068582185&kff1221=0.9132901&kff1222=0.35493663&kff1223=0.022978147&k10=4.1&k3c=509.5&k3e=338.2&kf=15.0&k5=53.0&kff1267=333.656&kff1266=472.0&k1f=855.0&kff1268=138.991&kff1238=13.7&k42=14.628&kff1240=3.3341064&kff1001=25.813221&kd=28.0&kff1272=3.8811188&kff1263=15.0 HTTP/1.0" 200 466 "-" "-"
192.168.10.34 - - [02/Dec/2020:16:08:40 +0100] "GET /upload_data.php?eml=aldo@iiiiiiiiiii.it&v=8&session=1606914493645&id=0f607264fc6318a92b9e13c65db7cd3c&time=1606914574517&kff1005=8.950457696363827&kff1006=44.40110186089179&kff1001=0.8137804&kff1007=351.36087&ke=24.5&k4=34.901962&k43=12.54902&k7=-1.5625&k6=-7.8125&kff1237=1.9242544&kff120c=16.269817&k21=0.0&k31=28732.0&k2c=0.0&kc=1404.5&kff1239=4.0&kff1010=51.76384191960123&kff1006=44.40110186089179&kff1005=8.950457696363827&kff123b=351.36087&kff123a=17.0&k24=1.0558472&k34=1.0102844&kff1215=0.835&kff1298=38.413773&kff1297=0.0&kff1296=61.586227&k11=18.82353&k47=49.019608&k45=0.0&kff1270=1008.5686&k33=99.0&kb=26.0&kff1202=-10.587755&kff124d=14.7&kff1249=14.851181&k44=1.0&kff1226=0.1169237&kff1273=0.08718999&kff1220=0.0068338118&kff1221=0.9018191&kff1222=0.37885493&kff1223=0.021439651&k10=4.57&k3c=555.7&k3e=420.7&kf=11.0&k5=54.0&kff1267=30.436&kff1266=79.0&k1f=464.0&kff1268=48.796&kff1238=13.8&k42=14.628&kff1240=3.5587158&kff1001=0.8137804&kd=12.0&kff1272=16.0&kff1263=24.0 HTTP/1.0" 200 466 "-" "-"
192.168.10.34 - - [02/Dec/2020:16:08:40 +0100] "GET /upload_data.php?eml=aldo@iiiiiiiiiii.it&v=8&session=1606914493645&id=0f607264fc6318a92b9e13c65db7cd3c&time=1606914778517&kff1005=8.950370924186352&kff1006=44.40112287961054&kff1001=0.060286738&kff1007=289.86005&ke=26.5&k4=38.431374&k43=21.960785&k7=-1.5625&k6=-3.90625&kff1237=0.060286738&kff1204=0.007081389&kff120c=16.2769&k21=0.0&k31=28732.0&kff1269=57.0&k2c=0.0&kc=1251.5&kff1239=4.0&kff1010=51.570902522895906&kff1006=44.40112287961054&kff1005=8.950370924186352&kff123b=289.86005&kff123a=14.0&k24=1.0012512&k34=1.0003052&kff1215=0.71999997&kff1298=82.786&kff1297=0.0&kff1296=17.213997&k11=18.431374&k47=50.980392&k45=2.3529413&kff1270=1008.5029&k33=99.0&kb=37.0&kff1202=-8.99234&kff124d=14.7&kff1249=14.718393&k44=1.0&kff1220=0.054426428&kff1221=0.90425974&kff1222=0.33370298&kff1223=0.0047066933&k10=6.76&k3c=494.09998&k3e=347.6&kf=14.0&k5=51.0&kff1267=222.028&kff1266=283.0&k1f=671.0&kff1268=61.439&kff1238=13.7&k42=14.648&kff1240=3.2999268&kff1001=0.060286738&kd=0.0&kff1272=0.87323946&kff1263=10.333333 HTTP/1.0" 200 466 "-" "-"
[... cut ...] f1215=0.835&kff1298=56.843647&kff1297=0.0&kff1296=43.156353&k11=23.529411&k47=63.52941&k45=12.156863&kff1270=1004.9956&k33=99.0&kb=88.0&kff1202=-1.5954151&kff124d=14.7&kff1249=14.687887&k44=1.0&kff1220=-0.02440647&kff1221=0.8559349&kff1222=0.47135547&kff1223=0.029753128&k10=21.78&k3c=617.0&k3e=431.9&kf=14.0&k5=60.0&kff1267=399.962&kff1266=793.0&k1f=1171.0&kff1268=393.65&kff1238=13.7&k42=14.628&kff1240=3.3096924&kff1001=28.835953&kd=28.0&kff1272=11.627907&kff1263=23.4375 HTTP/1.0" 200 466 "-" "-"

aldodemi, 2020-12-02:
About LAMP server hypothetical problem let me say that, on the same server, I've a running Owncloud installation (last version) that work without problem.

aldodemi, 2020-12-02:
Now I've also installed in another server where torque is the only LAMP application, same behavior

aldodemi, 2020-12-02:
[image]

aldodemi, 2020-12-02:
Apache error log in the new server when excecute http://ns2server/upload_data.php:

[Wed Dec 02 17:40:34.436884 2020] [mpm_prefork:notice] [pid 47423] AH00163: Apache/2.4.46 (Ubuntu) configured -- resuming normal operations
[Wed Dec 02 17:40:34.436956 2020] [core:notice] [pid 47423] AH00094: Command line: '/usr/sbin/apache2'
[Wed Dec 02 17:40:41.379043 2020] [php7:notice] [pid 47424] [client 192.168.10.35:56441] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 8

surfrock66, 2020-12-02:
I'm checking now, somehow the table name isn't getting defined.  Taking out the password, can you show me what's in creds.php?

surfrock66, 2020-12-02:
actually strike that, can you go into mysql and run this, replacing the $db_name and $db_table with the ones form creds.php:

SELECT table_name FROM INFORMATION_SCHEMA.tables WHERE table_schema = '$db_name' and table_name like '$db_table%' ORDER BY table_name DESC LIMIT 1;

aldodemi, 2020-12-02:
<?php

//echo "\r\n";
// MySQL Credentials
$db_host = 'localhost';
$db_user = 'torqueuser';     // Enter your MySQL username
$db_pass = 'xyzxyz';     // Enter your MySQL password
$db_port  = '3306';    // Enter yout MySql PORT
$db_name = 'torque';
$db_table = 'raw_logs';
$db_keys_table = 'torque_keys';
$db_sessions_table = 'sessions';
$gmapsApiKey = 'AIzaSyAHyCSYi_VZpn3AYLWIKluI3MQzcVKjXOM'; // OPTIONAL: Create a key at https://developers.google.com/maps/documentation/javascript/

// Array of user credentials for Browser login
$users = array();
// $users[] = array("user" => "torque", "pass" => "open");      // Sample: 'torque' / 'open'
// $users[] = array("user" => "second", "pass" => "mypass");    // Add additional strings for more users

//If you want to restrict access to upload_data.php,
// either enter your torque ID as shown in the torque app,
// or enter the hashed ID as it can found in the uploaded data.
//The hash is simply MD5(ID).
//Leave empty to allow any torque app to upload data to this server.
$torque_id = '';        //Sample: 123456789012345
$torque_id_hash = '';   //Sample: 58b9b9268acaef64ac6a80b0543357e6
//Just 'settings', could be moved to a config file later.
$source_is_fahrenheit = false;
$use_fahrenheit = false;

$source_is_miles = false;
$use_miles = false;

$hide_empty_variables = true;
$show_session_length = true;

#Sessions less than limit will not be shown
$min_session_size = 20;

//echo "\r\n";
?>

aldodemi, 2020-12-02:
mysql> SELECT table_name FROM INFORMATION_SCHEMA.tables WHERE table_schema = 'torque' and table_name like 'raw_logs' ORDER BY table_name DESC LIMIT 1;
+------------+
| TABLE_NAME |
+------------+
| raw_logs   |
+------------+
1 row in set (0.00 sec)

mysql>

surfrock66, 2020-12-02:
Can you do it again but add a % after "raw_logs" so it looks like "raw_logs%"?  I wonder if since this is a new instance if there's some problem creating the first monthly table.

aldodemi, 2020-12-02:
mysql> SELECT table_name FROM INFORMATION_SCHEMA.tables WHERE table_schema = 'torque' and table_name like 'raw_logs%' ORDER BY table_name DESC LIMIT 1;
+------------+
| TABLE_NAME |
+------------+
| raw_logs   |
+------------+
1 row in set (0.00 sec)

surfrock66, 2020-12-02:
So, the code says "Here's the current year month, so we want to write to a table for "raw_logs_YYYY_MM" and to do that, it copies from the previous month's table, and if it finds no previous month, it copies from raw_logs.  You have the raw_logs table which is good, but what's failing is this:

`
$newest_table_list = mysqli_query($con, "SELECT table_name FROM INFORMATION_SCHEMA.tables WHERE table_schema = '$db_name' and table_name like '$db_table%' ORDER BY table_name DESC LIMIT 1;");
$newest_table = "";
while( $row = mysqli_fetch_assoc($newest_table_list) ) {
  $newest_table = $row["table_name"];
}`

You ran that query and got a value back for table_name, but this isn't finding anything like that in the result.  In your local copy of data_upload.php can you update line 8 and change "table_name" to "TABLE_NAME"?  I don't think case matters (it doesn't matter to me) but it's a shot.

surfrock66, 2020-12-02:
Actually if there's an issue with the whole mysql connection I guess it could be the same error; can you clear your error logs for apache and mysql and just try to load the sessions page and see if anything pops up in the logs (specifically, is there a mysql connection error when it even tries to read sessions)?

aldodemi, 2020-12-02:
Changing "table_name" to "TABLE_NAME" see solve the problem!
[image]
Tommorow morning I'll try form the app

surfrock66, 2020-12-02:
That is so weird; I'll have to make a patch for that.  I have a few small patches to make, I'll update the code, thanks for the report!

surfrock66, 2020-12-02:
Fixed in https://github.com/surfrock66/torque/commit/0abf8a5ac9924b9a5d6aec9cb93738623009f6ae

closed 2020-12-02 by surfrock66