surfrock66/torque #37: need help please on upload_data.php

zhi0oo, 2021-01-10:
needd help please

the problem is on default upload_data
status is 
`You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near '' at line 1`

this is done with OK if remove line "OR DIE"
the status upload_data is OK but can't upload from torque data

`PHP Warning:  in_array() expects parameter 2 to be array, null given in `

then remove 

`if (mysqli_num_rows($result) > 0) {
  while ($row = mysqli_fetch_assoc($result)) {
    $dbfields[]=($row['Field']);
`

all OK but on session become

`Table 'otorque.raw_logs_2021_01' doesn't `exist``

thank  so much for all your help
this nice repo 👍  goodjob

surfrock66, 2021-01-12:
Weird, what's your php version?

zhi0oo, 2021-01-12:
``PHP 7.2.34-8+ubuntu16.04.1+deb.sury.org+1 (cli) `

zhi0oo, 2021-01-12:
Any suggestions that need to be installed on a fresh server
and details the version

thanks

surfrock66, 2021-01-13:
It's taking me a bit to get my head back into this code (I haven't looked at it in like a year as I haven't even been driving since most of COVID) but just off the bad, that warning indicates the "in_array()" function on line 106, but that code is line 100 in the master branch: https://github.com/surfrock66/torque/blob/master/web/upload_data.php 
That makes me wonder if you're using the most up to date commits?  I don't see a line number in the actual first MySQL error so I'm not sure which part of upload is failing,   Which line did you remove the "OR Die" from?

zhi0oo, 2021-01-14:
all 10 syntax

 `or die(mysqli_error($con))`

on /upload_data.php
 
no error and some data success uploaded to database

 if only remove at line 11,  no error but only loading on torque

thank

surfrock66, 2021-01-16:
I ran through the code again, can you try stepping through the database creation code in the php in mysql?  I know that's vague, but I don't understand where the failure is.  Part of upload data is a process of this logic:
1) Is there a table for this year/month?
2) If so, upload data to it
3) If not, make it

I suspect the issue is there, and tables aren't being made.

zhi0oo, 2021-01-16:
yess right there no table
how to make the table year month

are like this
`  `Y` varchar(15) NOT NULL,
  `m` varchar(15) NOT NULL,
`

zhi0oo, 2021-01-17:
i have upload year mount table
but no data was show on session

oisac, 2021-07-03:
I get the same error after the change in upload_data.php on line 8.

$newest_table = $row["TABLE_NAME"]; results in the above error.
If I change it back to "table_name" (lowercase) the error goes away and I get "OK!" and data is uploaded.

I'm using PHP Version 7.4.15.

surfrock66, 2021-07-03:
Ugh my bad, this is a regression, I had my source out of sync and edited the wrong local copy.  I just pushed a fix.  Thanks for the report.

closed 2021-07-03 by surfrock66