surfrock66/torque #45: MariaDB Error

itschrisonline, 2021-08-11:
I am getting this error on nginx and mariadb

You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near 'WHERE ( CONCAT(YEAR(FROM_UNIXTIME(session/1000)), '_', DATE_FORMAT(FROM_UNIXT...' at line 1

Any help would be great

surfrock66, 2021-08-13:
I've been on vacation, sorry for the delay.

There's not enough there for me to debug; can you confirm the rest of that debug line, and can you confirm that there is a session in the table?

mo5245, 2021-09-01:
I've been running Nginx and MariaDB (10.4) for a while now, more info on where the error is received and the MariaDB & PHP version you're running would help. I've done fresh VM test install a while back to check for errors if DB is empty on new install, only has minimal js and php vars undefined.