surfrock66/torque #46: No more load data

aldodemi, 2021-08-30:
Server Ubuntu 21.04 - hirsute VERSION="21.04 (Hirsute Hippo)"
mysql  Ver 8.0.26-0ubuntu0.21.04.3 for Linux on x86_64 ((Ubuntu))

No data loaded anymore: if connect via web to "upload_data.php" page it responds "OK!"
but torque PRO remain in "loading..."

Found this in apacha2 error log:
"[php7:notice] [pid 2934453] [client zz.kk.xx.yy:47808] PHP Notice:  Undefined index: table_name in /var/www/torque/upload_data.php on line 115"

Thanks 
Aldo

genius3000, 2021-08-30:
This sounds like another instance of the issue found in #36. Try changing table_name on line 115 of upload_data.php to TABLE_NAME (lowercase to uppercase).

surfrock66, 2021-08-30:
Yea, this is the weirdest thing.  I did a new grep and found another few I missed, and pushed a new commit.  I hope this is all of it now, I did a recursive grep on "table_name" in the whole repo.

aldodemi, 2021-08-31:
This problem seems solved, thank you

closed 2021-08-31 by aldodemi