surfrock66/torque #47: Error during data load

aldodemi, 2021-08-31:
Server Ubuntu 21.04 - hirsute VERSION="21.04 (Hirsute Hippo)"
mysql Ver 8.0.26-0ubuntu0.21.04.3 for Linux on x86_64 ((Ubuntu))

[Tue Aug 31 11:49:36.209065 2021] [php7:notice] [pid 3305314] [client xx.yy.zz.154:48513] PHP Notice:  Undefined variable: sessTimeStart in /var/www/torque/upload_data.php on line 197
[Tue Aug 31 11:49:36.209091 2021] [php7:notice] [pid 3305314] [client xx.yy.zz.154:48513] PHP Notice:  Undefined variable: sessTimeEnd in /var/www/torque/upload_data.php on line 201
[Tue Aug 31 11:49:36.209095 2021] [php7:notice] [pid 3305314] [client xx.yy.zz.154:48513] PHP Notice:  Undefined variable: sessSize in /var/www/torque/upload_data.php on line 205
[Tue Aug 31 11:49:36.209100 2021] [php7:notice] [pid 3305314] [client xx.yy.zz.154:48513] PHP Notice:  Undefined variable: sessTimeStart in /var/www/torque/upload_data.php on line 207

Thank you
Aldo

surfrock66, 2021-09-01:
I don't know how to debug this exactly; the code starting on line 161 of upload_data is checking if a session already exists in the DB for your current session.  If it doesn't exist, create one from the current datapoint being uploaded,  if it does exist, use variables pulled from the sessions table.  It looks like it's thinking a session exists, but isn't getting variables back.  I can't tell if there's an empty session getting created, or if it's trying to read a session and skipping that code check?

What would be helpful would be to enable access logging, then make a new session on your next (short) drive, then look at all access logs to upload_data to see what was in the get request, then you'd get a session ID and can look in the DB for whats in there.

aldodemi, 2021-09-02:
I'm sorry, it was my fault, my be probably not correctly updated web directory.
Reloaded from scrath no more errors.
Many thanks.
Aldo

closed 2021-09-02 by aldodemi