surfrock66/torque #55: Fresh install css fails to load

oisac, 2023-02-23:
I cloned the repo on a fresh linode instance and no formatting or css is loading. I also run this repo at Dreamhost with no issues at all. Is there something that needs to be configured for PHP on linode that I'm missing?

oisac, 2023-02-24:
I did view source and all links go to the specified file. I have no idea what the issue is. I'm going to try reinstalling apache. Dreamhost is shared hosting so I just cloned and all was good, I don't have access to any apache/php configs.

oisac, 2023-02-24:
Nuked my server and started over. Looks like I missed something in the initial install even though other php stuff worked. All is good now. Issue resolved.

closed 2023-02-24 by oisac

surfrock66, 2023-02-25:
Glad it worked, enjoy!